pantor/ruckig #109: return_error_at_maximal_duration: how to change maximal duration?

ethindp, 2022-02-26:
The maximal duration is 7.6e3, or 7600. Is there a way to change this and does turning it off break anything? And why is it so low?

pantor, 2022-03-01:
This is to ensure numeric stability, it's also in the [paper](https://arxiv.org/pdf/2105.04830.pdf) on the top right of page 6. "Low" is definitely relative: Ruckig was designed for robotics and machining applications, so when using SI units this corresponds to time-optimal trajectories durations of over 2 hours. That might change if you have other requirements regarding the precision, for more information please see the [numerical stability section](https://github.com/pantor/ruckig#tests-and-numerical-stability).

In general, numerical stability is very unpredictable and depends a lot on the actual use case. Feel free to comment that check out and see if Ruckig still works for your application.

pantor, 2022-03-04:
Feel free to reopen if you have any more questions!

closed 2022-03-04 by pantor