pantor/ruckig #110: Velocity never reaches absolute maximum

ethindp, 2022-02-26:
So in my tests for a project that a friend and I have integrated this into, we've noticed that the velocity never reaches its absolute maximum. E.g.: if we set maximum acceleration to 10.0, 10.0, 10.0, the max jerk to 1.0, 1.0, 1.0, and the max velocity to 100.0, 100.0, 100.0, the velocity usually always hovers around 100.0, 99.7936806409023, 99.09316204859, but never goes to 100.0, 100.0, 100.0. I'm no expert on physics, though; is this a floating-point problem, a physics-based limitation, or something else? We're doing these tests in an attempt to figure out the best parameters for our use-case. We just set the current position/velocity/acceleration to the new position/velocity/acceleration, but I have a feeling we're missing something.

pantor, 2022-03-01:
Hi @ethindp,

I assume you're using Ruckig with 3 degrees of freedom. Are their current states, target states and kinematic limits all equal? If not, Ruckig slows down the non-limiting DoFs so that all 3 DoFs reach the target state at the exact same point of time. This is (when possible) done by reducing the maximum velocity. If you want to reach the target state as fast as possible for all DoFs independently, Ruckig has the `synchronization` parameter.

pantor, 2022-03-04:
Feel free to reopen if you have any more questions!

closed 2022-03-04 by pantor