pantor/ruckig #111: install ruckig via pip: fatal error: 'httplib/httplib.h' file not found

fferri, 2022-03-04:
I'm trying to install ruckig on macOS and Python 3.9:

```
python3 -m pip install pybind11
export pybind11_DIR=/usr/local/lib/python3.9/site-packages/pybind11
python3 -m pip install ruckig
```

last command fails with an error:

      /private/var/folders/73/362j5fts3d71myns0t7_kmd00000gn/T/pip-install-0txlv204/ruckig_98dcedb43575469faf81bb439ce2066b/include/ruckig/calculator_online.hpp:11:10: fatal error: 'httplib/httplib.h' file not found
      #include <httplib/httplib.h>
               ^~~~~~~~~~~~~~~~~~~
      1 error generated.
      1 error generated.
...

× Encountered error while trying to install package.
╰─> ruckig

pantor, 2022-03-07:
This should be fixed with release v0.6.5. Thanks for reporting!

closed 2022-03-07 by pantor

fferri, 2022-03-07:
Thanks, v0.6.5 via pip built and installed fine!