pantor/ruckig #114: Rust bindings

ethindp, 2022-03-10:
Hi there,
It would be really cool to see rust bindings for this project. Would such a thing be possible?

pantor, 2022-03-11:
I would love to have Rust bindings for Ruckig as well. It's currently not my main focus, as Rust is not that widely used in the robotics or automation community (yet), but maybe somebody else decides to give it a try in in the meantime?

jmpinit, 2022-05-02:
These are not proper bindings but might at least serve as a working starting point for others: https://github.com/jmpinit/rust-ruckig

pantor, 2022-05-02:
The repository seems to be private @jmpinit, but I'm excited!

jmpinit, 2022-05-02:
Whoops, sorry about that. It's public now.

petrikosk, 2023-11-12:
There is now a full Rust port of Ruckig here: https://github.com/petrikosk/rsruckig