pantor/ruckig #118: Add a method to reset the Ruckig object with new `DOF` and `delta_t`

AndyZe, 2022-04-08:
I'm trying to do something like this with every control loop iteration because the period of my control loop might change:

`smoother = ruckig::Ruckig<ruckig::DynamicDOFs>(new_DOF, new_delta_t));`

It would be nice if there were something like a `reset(size_t new_DOFs, double new_delta_t)` method so I didn't have to re-initialize like that:

`smoother.reset(new_DOF, new_delta_t);`

Here's an explanation on why that's inefficient:  https://stackoverflow.com/a/2530889/3499467

Danielxyz123, 2022-04-09:
Is it equal to #42 ?

AndyZe, 2022-04-09:
That sounds different to me. I want to run Ruckig for one point at a time, in realtime fashion. #42 was talking about running once, then sampling the solution afterwards. (I think...)

pantor, 2022-05-15:
Hey @AndyZe! `delta_t` should be straight-forward to change, in fact I don't see a problem with just removing `const` from `Ruckig::delta_time`. Would that already help in your case?

It's a bit harder for `DoFs`, I think we would need a proper reset function here.

AndyZe, 2022-05-15:
non-const `delta_t` sounds good :+1: You are right, I don't have any need to change `DoFs` in real time.

pantor, 2022-05-16:
`Ruckig::delta_time` is now non-const with the latest commit on the master branch.

I'll close this for now, as I think if one wants to change the `DoFs` of Ruckig efficiently in real-time, it'd look a bit like a code smell to me. If anybody comes up with a good use-case, I'll be happy to add a proper reset method though.

closed 2022-05-16 by pantor

AndyZe, 2022-06-18:
By the way, thank you! This should be merged into ros2_control soon and I think it will improve the performance of all ROS hardware a lot. Believe it or not, vel/accel limits are not currently checked by ros2_control in any way.

https://github.com/ros-controls/ros2_controllers/pull/324

AndyZe, 2022-06-18:
Would you be able to release Ruckig for ROS2 again? Specifically Rolling and Humble distributions.

I don't think `Ruckig::delta_time` is non-const in the current verison.

AndyZe, 2022-06-18:
It's not a huge deal if you can't.

pantor, 2022-07-29:
Hey @AndyZe, I just wanted to release Ruckig v0.7.1 for ROS via bloom. However there seems to be some issue related to 2FA, and it's also challenging to get right within CD. And I think the release repo was changed to https://github.com/ros2-gbp/ruckig-release recently. I'm assuming you know more about the ROS release process than I do. Any chance you could support me here or take this over?

AndyZe, 2022-07-29:
Let me put you in contact with somebody. @vatanaksoytezer do you or Adam have time to help with a release? I have @pantor's email, if you can

rhaschke, 2022-12-15:
> I think the release repo was changed to https://github.com/ros2-gbp/ruckig-release recently.

All release repos were automatically changed when preparing for Humble I think.
As you probably don't have write access to this repo yet, you can either:
- ask distro maintainers (e.g. @nuclearsandwich) for write access to the release repo
- switch back to your old release repo via bloom:
  ```
  bloom-release --ros-distro rolling --override-release-repository-url https://github.com/pantor/ruckig-release.git ruckig
  ```

rhaschke, 2022-12-15:
Other persons cannot help with the release process as one needs to have write access to the release repo.

pantor, 2022-12-16:
Good to know, @rhaschke. The main discussion for setting up new ROS releases is in #137. In short, we will give @vatanaksoytezer write access to the ros2-gbp/ruckig-release repo.