pantor/ruckig #121: runtime error in offline trajectory generation

nkashiri, 2022-04-28:
I occasionally get the following kind of error that crashed my program. Any idea what may cause this? 

RuntimeError: [ruckig] error in step 2 in dof: 26 for t sync: 6.603280 input:
inp.current_position = [0.5, -0.7, -0.7, -1.5, -0, -0.2, 0.5, 0.7, 0.7, -1.5, 0, -0.2, -0, 0, -0, 0, 0, 0, 0, -0, 0, -0, 0, 0, 0, -0, 0, -0, 0, 0, -0, 0, -0, 0, 0, 0, -0, 0]
inp.current_velocity = [2.689534009085704e-06, -2.689534009088848e-06, -2.689534009086246e-06, -2.689534009088414e-06, 2.689534009087655e-06, -2.689534009089241e-06, 2.68953400908874e-06, 2.689534009087113e-06, 2.689534009086246e-06, -2.689534009088848e-06, -2.689534009089065e-06, -2.689534009089228e-06, 2.689534009088198e-06, -2.68953400908668e-06, 2.689534009089715e-06, -2.689534009087113e-06, -2.4911206284336e-12, 0, -2.689534009089065e-06, 2.689534009089065e-06, -2.689534009087764e-06, 2.689534009059358e-06, -2.4911206284336e-12, 0, -2.689534009088957e-06, 2.689534009088631e-06, -2.689534009088198e-06, 2.689534009031602e-06, -2.4911206284336e-12, 0, 2.689534009088848e-06, -2.689534009088414e-06, 2.689534009086463e-06, -2.689534009142625e-06, -2.4911206284336e-12, 0, -2.689534009086869e-06, -2.689534009086951e-06]
inp.current_acceleration = [-0.001639979880695402, 0.001639979880696266, 0.001639979880695815, 0.00163997988069603, -0.00163997988069612, 0.001639979880696489, -0.001639979880696373, -0.001639979880695849, -0.001639979880695794, 0.001639979880696522, 0.001639979880696495, 0.001639979880696486, -0.00163997988069612, 0.001639979880695946, -0.001639979880696397, 0.001639979880696085, 0, 0, 0.001639979880696328, -0.001639979880696418, 0.001639979880696141, -0.001639979880696363, 0, 0, 0.001639979880696443, -0.001639979880696245, 0.001639979880696175, -0.001639979880695752, 0, 0, -0.001639979880696356, 0.001639979880696342, -0.001639979880695766, 0.001639979880696307, 0, 0, 0.001639979880695763, 0.001639979880695815]
inp.target_position = [1.5, -0.7, -0.7, -1.5, 0, -0.2, 1.5, 0.7, 0.7, -1.5, 0, -0.2, 0, 1.7, 2.3, 0.6, 0, 0, 0, -1.7, -2.3, -0.6, 0, 0, 0, -1.7, -2.3, -0.6, 0, 0, 0, 1.7, 2.3, 0.6, 0, 0, 0, 0]
inp.target_velocity = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
inp.target_acceleration = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
inp.max_velocity = [0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5]
inp.max_acceleration = [0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5]
inp.max_jerk = [0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5, 0.5]

pantor, 2022-05-18:
Hey @nkashiri, I've committed a fix for this to the master branch. Can you confirm that it solves your issue?

nkashiri, 2022-05-24:
Thanks @pantor 
As it is not a repeatable error (it happens sometimes), I cannot confirm immediately. I'll update this issue

closed 2022-06-22 by pantor