pantor/ruckig #122: what  does jf means in the corresponding paper step 2 Time Synchronization?

oridong, 2022-05-02:
What does the **_jf_**  mean? it is only for the last segment instead of every interval in seven segments ?

pantor, 2022-05-03:
In the paper, _j_f_ stands for the absolute jerk value of every non-zero segment (so up to four out of the seven segments).

oridong, 2022-05-05:
could you give me some details about how to solve S2?

$$\mathcal S2 :(T_p,p_0,p_f,v_0,v_f,a_0,a_f,v_{max},a_{max},a_{min},j_{max}) \mapsto (t_1,t_2,t_3,t_4,t_5,t_6,t_7,j_f) $$

I have read **ruckig-step2-uddu.nb** and I have guessed some facts.

In some cases, **vmax** should be changed, in other cases, **jmax** should be changed indicated by **Join[tVars, {vMax}]** or **Join[tVars, {jMax}]** in **Solve[]**

> Information
> - (1) Case 1a: Acc0_Acc1_Vel Solution 1 
> - (2) Case 2a: Acc0_Acc1 Solution 1 
> - (3) Case 2b: Acc0_Acc1 (a0=0, af=0) Solution 1

In these cases (1a,2a,2b), analytical roots can be obtained, so the expression of roots can be assigned to **profile.t ([0]~[6])**

> - (4) Case 3a: Acc1_Vel, Root t1
> - (5) Case 4: Acc0_Vel, Root t5
> - (6) Case 5: Vel, Root t1

In these case (3a,4,5), no analytical root by mathematica, so numerical method should be adopted which is implemented in **solveQuartMonic** function

>With acceleration Plateau -> a3 != 0
resultTa = {
   Solve[{a7 == af, v7 == vf, p7 == pf, a3 == aPlat, t2 == 0, t6 == 0,
      t7 == 0, tAll == tf}, ...],
    ....
   Solve[{a7 == af, v7 == vf, p7 == pf, a3 == aPlat, t1 == 0, t2 == 0,
      a5 == aMax, tAll == tf}, ...]
   };

does this indicate try to adapt the amax?

>Condition for maximal allowed target acceleration

where this result is showed in **position-step2.cpp**?

I wish you give some systematic and overall explanation about how to solve S2 in your free time.
In my opinon,S2 is the inverse version of S1 with respect to  the complete duration and  inverse is usually complex.

pantor, 2022-05-16:
In general, Ruckig tries to minimize v_max (corresponding to minimizing kinetic energy). That's however not always possible in a simple and fast way, and then we switch to usually minimizing j_max. 

The cited example does not show an adaption of a_max, this corresponds to an acceleration plateau of a3 != 0. One can show that these profiles are necessary for a general solution for arbitrary states and durations.

The condition for a maximal allows target acceleration would e.g. be `a5 == aMax` or `a5 == aMin`.

S2 is not really an inverse version of S1. S2 takes the duration into account, and therefore needs one more degree of freedom in return to make it algebraically solvable. This could be v_max, a_max, j_max or even by using completely different profiles. For example, the jerk doesn't even need to be constant over time. S2 is much more flexible, however you're right that it is mathematically more complex to solve. So there is no right or wrong here, but Ruckig tries to optimize S2 for: (1) minimum energy consumption (or minimum maximum velocity), (2) computational complexity, and (3) motion smoothness.

closed 2022-05-16 by pantor