pantor/ruckig #126: Is there a way to customize the position curve (e.g. with cubic Bezier curve control points)?

bobbradley, 2022-05-13:
The generated trajectory for position seems to follow a standard S curve (ease-in-ease-out). Is there a way to change the way this position curve is generated while still restricting to the velocity/acceleration/jerk limits?

For example, if I wanted to generate a curve that looks like this:
<https://cubic-bezier.com/#.99,.06,.75,.94>

I can perform the bezier curve calculation part to generate a raw position, but I wasn't sure how to use that with ruckig. I tried updating the target_position after each step of the control loop, but that caused the the output time to reset each time.

pantor, 2022-05-15:
Hi @bobbradley. No, there is no easy way to use Bézier curves. Ruckig calculates time-optimal trajectories, and S-curves are the only solution for this.

Why do you want to use Bézier curves anyway? If you really want to keep them, you can use them as your geometric path `r(s)` and generate a 1-DoF curve with Ruckig to output the path parameter `s`. However, you would need to calculate the upper bounds of the velocity/acceleration/jerk limits yourself.

bobbradley, 2022-05-15:
Thanks for the response. The reason I wanted to use Bézier curves is to allow the trajectory to be tuned to what looks or feels the best. Graphic user interfaces do this to make transitions more aesthetically pleasing. This would apply that technique to physical movement. For example, moving more slowly at the end than the beginning.

It doesn't specifically need to use Bézier curves though. I mainly just wanted more control over the S-curve. Splitting the trajectory into multiple segments could possibly emulate it. For example, change target_position and minimum_duration or max_velocity/max_acceleration before the beginning of each segment.

closed 2022-05-16 by pantor