pantor/ruckig #130: ErrorExecutionTimeCalculation

aseligmann-ipu, 2022-06-07:
Hi @pantor,

Thanks for the great work on Ruckig :+1: 

I'm running into an issue where Ruckig often throws an `ErrorExecutionTimeCalculation` in a mostly reproducable scenario (decelerating one joint while accelerating another).

The target is updated often, as are the joint limits, so the current state is not always valid. 
However, I ensure that the target state is valid and reachable.

Here's the error printout: 
```
[ruckig] error in step 1, dof: 3 input: 
inp.current_position = [0.1720894642466409, 3.62255615705579, 1.675754445058354, 2.181127634157466, 0.7829547024446563, 0]
inp.current_velocity = [0, -0.06422550797049381, -0.006910864193129071, 0.1207111377965507, 0, 0]
inp.current_acceleration = [0, 0, 0, -0.3999999999978001, 0, 0]
inp.target_position = [0.1703441349946466, 3.587689540855463, 1.671344092861895, 2.271916653384632, 0.7826056365942573, -0.003141592653589793]
inp.target_velocity = [0, -0.06423623454575711, -0.007006078877926925, 0.1187909578397501, 0, 0]
inp.target_acceleration = [0, 0, 0, 0, 0, 0]
inp.max_velocity = [0.06981316999999999, 0.06423623454575711, 0.007006078877926925, 0.1187909578397501, 6.981317008, 34.90658504]
inp.max_acceleration = [0.034906585, 0.261799388, 0.043633231, 0.4, 4.01425728, 34.90658504]
inp.max_jerk = [100, 100, 100, 100, 100, 100]
```

For reference, the input is initalised as
```
inp.enabled.at(0) = false;
inp.enabled.at(4) = false;
inp.enabled.at(5) = false;
inp.control_interface = ruckig::ControlInterface::Position;
inp.synchronization = ruckig::Synchronization::None;
inp.duration_discretization = ruckig::DurationDiscretization::Continuous;
```

Any help is much appreciated!

aseligmann-ipu, 2022-08-23:
Hi @pantor,
Do you have any expectation for when this bug might be fixed?
Is there any additional information I can provide?

pantor, 2022-08-28:
Hey! This should be fixed now with the latest commit on the master branch.
Can you confirm this?

aseligmann-ipu, 2022-08-30:
I have not observed this since the latest commit, so it seems to have been fixed.
Thank you :champagne:

closed 2022-08-30 by aseligmann-ipu