pantor/ruckig #134: Question about combining this with other motion planning libraries

ethindp, 2022-06-30:
@pantor I'm considering combining this with [RVO2-3D](https://github.com/snape/RVO2-3D). RVO2-3D is a three-dimensional adaptation of [RVO2](https://github.com/snape/RVO2), which uses [Optimal Reciprocal Collision Avoidance](https://gamma.cs.unc.edu/RVO2) (ORCA). The reason I want to combine these two libraries is that this library only provides trajectory planning, and doesn't provide collision avoidance with other objects in a 3D space. ORCA provides that collision avoidance mechanism and eliminates the need to implement algorithms such as bounding boxes or ray-casting where it isn't necessary. However, I am unsure how these two libraries would fit nicely together: both provide coordinate sets while an object is in motion, and so I would need to identify a way of informing Ruckig of the information that RVO2 provides in real-time so that Ruckig could recompute the trajectory if necessary. Is combining these two libraries feasible or a good idea?

pantor, 2022-07-09:
Hi @ethindp,

I'm not really an expert for RVO2 (I just learned about it). It seems that it just includes a soft velocity constraint. If you want to add acceleration and jerk constraints, it might get a bit tricky. Ideally, you would calculate an initial trajectory with RVO2, smooth the trajectory with Ruckig, and then inform RVO2 about the smoothed positions and velocities. I'm not sure if RVO2 is able to handle updated positions at every iteration. The smoothing of a given trajectory could be done similar to [this MoveIt PR](https://github.com/ros-planning/moveit2/pull/571). In the end, I'm not really sure if this is feasible or not - I just know one half of the story here.

Do you have any updates in the meantime?

ethindp, 2022-07-16:
@pantor I don't, unfortunately. I'd love to combine RVO2 with Ruckig since ruckig does not take into account agents around the actual trajectory. The reason I'd like to use RVO2 is because I'm not a mathematics expert and I'm unsure of the best way of devising an algorithm capable of performing N-dimensional agent detection and avoidance. Though RVO2 only covers the 2-and 3-dimensional cases, I figure that figuring out how to combine the two libraries by (as you noted) feeding the data of one of them into the other is probably the wisest corse of action. The problem is, of course, identifying which library should ultimately have control. RVO2 does allow the dynamic updating of the simulation -- that's how you use it -- but I don't know if its as granular as Ruckig allows you to be. Perhaps this is worth investigating and documenting?

closed 2022-07-25 by pantor

ethindp, 2022-07-25:
@pantor I see that this issue was closed as `completed`... Did you find a solution?

ethindp, 2022-09-16:
@pantor I've looked at the PR you linked to, and I'm confused on how I can apply that strategy to our (my friends and my) project. As in, I'm confused on how to go about smoothing the trajectory with Ruckig while allowing RVO2 to generally hold the reigns, as it were. We're not really super knowledgeable about robotic navigation and such, so we'd appreciate some tips or explanations on how we can get this to work.