pantor/ruckig #135: [Compile Error]: ‘af_scale’ may be used uninitialized in this function [-Werror=maybe-uninitialized]

MingshanHe, 2022-07-11:
Dear,

I have a compile error during `make` process.

The terminal information has been followed as:
[image]

I have tried to delete the `af_scale` variable and the project has been compiled successfully.
But this variable is the same as the others, like `v0_scale`, `a0_scale`, `vf_scale`. So I open this issue to get your help.

Best regards,
Mingshan

pantor, 2022-07-13:
This is a compiler warnings that is treated as an error due to the used `-Werror` compiler flag. The recent commit on the master branch removes that flag, so you should be able to compile Ruckig without error now. I'll keep that issue open until I fix the underlying warning. Thanks!

pantor, 2022-08-03:
The warning is now fixed on master.

closed 2022-08-03 by pantor