pantor/ruckig #142: Is it correct result?

tjdalsckd, 2022-10-17:
Thanks for the good paper.

I think
[image]

input parameter is

      input.current_position = {0};
      input.current_velocity = {20};
      input.current_acceleration = {150};
      
      input.target_position = {1};
      input.target_velocity = {10};
      input.target_acceleration = {207.154987413215};
      
      input.max_velocity = {70};
      input.min_velocity = {-70};
      input.max_acceleration = {500};
      input.min_acceleration = {7900};
      input.max_jerk = {2458000};

but the result is 
     a6 = -7893.536510410606
     max(v) = 35.341000017262
Substituting into the above formula,
the RHS is 62,307,918.64118525,
the LHS is  62,309,634.636835076

The left and right sides are different.

Is it correct trajectory?

[image]

pantor, 2022-10-17:
Hi @tjdalsckd! The trajectory seems to be correct and time-optimal to me. Or why do you think otherwise?

Where does the equation come from? If t6 is referring to the 6th step in the profile, then `t6 != 0` in this case. In particular, the equation depends on the maximum acceleration in your example.

tjdalsckd, 2022-10-17:
I used the following formula

https://github.com/pantor/ruckig/blob/612eb3cb6bc8699c48cf49170d136dfedb469ab2/src/ruckig/position-step1.cpp#L31

amin = 7900
vf = 10
af = 207.154987413215
j = 2458000

[image]

but real vmax is  35.341000017262.

pantor, 2022-10-21:
Your calculated trajectory has following times:
```
t1 = 0.000142392
t2= 0.0304877
t3 = 0.00341483
t4 = 0
t5 = 0
t6 = 0
t7 = 0.00329569
```
In particular, the minimum acceleration of the profile is `a6=-7893.63` and does not reach the acceleration constraint. Therefore, `t6` is zero. Furthermore, the equation doesn't correspond to your profile, and that's why you have the discrepancy between the two `v_max`.

closed 2022-10-21 by pantor