pantor/ruckig #147: Invalid input error, but input seems fine

AndyZe, 2022-12-14:
> [ERROR] - /seg_0/pick_bot: [1671034050.415487924] Ruckig jerk-limited smoothing failed with code: -100   (Invalid input)

But I don't see what about this is invalid:
```
[ WARN] - /seg_0/pick_bot: [1671034050.416114726] 0.00769231
Ruckig input:

inp.current_position = [-1.135053174917025, 2.79, 2.45, -0.58, -1.72, 1.015086525357003]
inp.current_velocity = [0, 0, 0, 0, 0, 0]
inp.current_acceleration = [0, 0, 0, 0, 0, 0]
inp.target_position = [-1.271207372084076, 2.79, 2.45, -0.5800000000000002, -1.72, 1.151216170589311]
inp.target_velocity = [-1.711010717370143, 0, 0, -2.054215846043843e-15, 0, 1.710702180253011]
inp.target_acceleration = [29.99999999998483, 0, 0, 3.601758583722072e-14, 0, -29.99459026559806]
inp.max_velocity = [3.15, 3.15, 3.15, 4.24, 4.75, 4.24]
inp.max_acceleration = [30, 30, 30, 30, 30, 30]
inp.max_jerk = [100, 100, 100, 100, 100, 100]
```

AndyZe, 2022-12-14:
Do we just need to make a new release for ROS1? I can grab somebody to do that if you give them admin access to pantor/ruckig and pantor/ruckig-release, I bet

gavanderhoorn, 2022-12-14:
You wouldn't need admin access to this repository. Just push access to the release repository.

Provided the necessary tags are created on `pantor/ruckig`.

Technically you could even do a "rogue" release, with a different release repository.

pantor, 2022-12-14:
I'd be happy to give someone write or maintain permissions to this repository for that 👍.  Let's keep the discussion for that in #137.

pantor, 2022-12-14:
Regarding the invalid input, keep in mind that the target acceleration needs to fulfill
```
target_acceleration <= sqrt(2 * max_jerk * (max_velocity - abs(velocity)))
```
so that the trajectory is able reach the target acceleration with limited jerk inside the velocity limits. The Readme has more information about it [here](https://github.com/pantor/ruckig#input-validation). 

It seems (at least) the first axis is violating that condition.

closed 2022-12-14 by AndyZe