pantor/ruckig #150: how i can use the lib in cartesian space.

chinahuangyong, 2023-01-10:


pantor, 2023-01-11:
Ruckig can be used flexibly for any underlying space with any number of degrees of freedom. For generating motions in Cartesian space, you need to find a representation that then can be used as input for Ruckig. There exist quite a few representations, from Euler angles to axis-angle to quaternion representations, each with different pros and cons.

As a rule of thumb, we recommend an [axis-angle representation](https://en.wikipedia.org/wiki/Axis–angle_representation) for Cartesian poses. In particular it allows to find a fixed axis between a start and a goal pose, and then the actual motion can be parametrized by three translation DoF and a single rotational DoF (the angle). This is very easy to implement, however only works if the start and goal velocities are zero.

closed 2023-01-11 by pantor

chinahuangyong, 2023-01-11:
can i get some examples?

chinahuangyong, 2023-01-11:
could you please give me some examples for reference?

pantor, 2023-01-11:
You can have a look at the [frankx repository](https://github.com/pantor/frankx).