pantor/ruckig #151: pip installation pybind11 issue

slartz42, 2023-01-11:
Hello, 

I'm hoping to use Ruckig for a project I'm working on but I can't seem to get it installed, I'm getting the following error below:

 CMake Error at CMakeLists.txt:82 (find_package):
        By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
        asked CMake to find a package configuration file provided by "pybind11",
        but CMake did not find one.
     
        Could not find a package configuration file provided by "pybind11"
        (requested version 2.6) with any of the following names:
     
          pybind11Config.cmake
          pybind11-config.cmake
     
        Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
        "pybind11_DIR" to a directory containing one of the above files.  If
        "pybind11" provides a separate development package or SDK, be sure it has
        been installed.

I ran `pip install pybind11` and it installed successfully, but I still get the error. I also ran `pip install cmake` which installed but did not fix the error.

I'm currently using a virtual environment in PyCharm, any help would be appreciated

pantor, 2023-01-12:
First, you can also install Ruckig for Python via `pip install ruckig`. 

If you need the latest version from the master branch, you can compile the Python package yourself. The easiest way to install the pybind11 dependency for that is to install it via a system-wide package manager (e.g. on ubuntu -> `apt install pybind11-dev`, on mac `brew install pybind11`). More info from pybind11 [here](https://pybind11.readthedocs.io/en/stable/installing.html) and [here](https://repology.org/project/python:pybind11/versions).

slartz42, 2023-01-12:
So you copy and paste the project description which I already read then closed out the issue. Did you even read my question?

What I got from this response is "pip install doesn't work and we don't care"

pantor, 2023-01-12:
Ah okay, I thought you were building the C++ project with cmake yourself and then only installing pybind11 with pip. That's why I gave some links to the pybind11 manual.

We provide pre-built Python packages for the major environments, and this issue might pop up when pip wants to build directly from source instead.

Then a couple of questions:
- Which Python version?
- Which OS?
- Which CPU architecture?

slartz42, 2023-01-12:
No problem, I probably should have chosen a different title. 

- Python 3.11
- Windows 10
- 64 bit processor

pantor, 2023-01-13:
We've now released v0.9.2 with a pre-built package for Python 3.11. Can you try again to run `pip install ruckig`?

slartz42, 2023-01-13:
That worked on the first try!! Thanks!

closed 2023-01-13 by pantor