pantor/ruckig #153: How to use ruckig::Trajectory with unknown DOF at compile time

AndyZe, 2023-03-17:
I was playing with the offline trajectory planner, like in this example:  https://github.com/pantor/ruckig/blob/master/examples/2_position_offline.cpp

It works great if the DOF can be hard-coded like this:

```
Trajectory<3> trajectory;
...
std::array<double, 3> new_position, new_velocity, new_acceleration;
trajectory.at_time(new_time, new_position, new_velocity, new_acceleration);
```

But with MoveIt, we don't know the num DOF until runtime.

This compiles, I'm not sure it will run, though:

`ruckig::Trajectory<ruckig::DynamicDOFs, ruckig::StandardVector> ruckig_trajectory(ruckig::DynamicDOFs);`

And I struggle with what to do here, since arrays can't be dynamically sized:

```
size_t num_dof = 6; // for example
std::array<double, num_dof> new_position, new_velocity, new_acceleration;
```

AndyZe, 2023-03-17:
Vectors instead of arrays actually do compile. Again, I don't know if this will run:

```
        std::vector<double> new_position;
        std::vector<double> new_velocity;
        std::vector<double> new_acceleration;
        ruckig_trajectory.at_time(time_from_start, new_position, new_velocity, new_acceleration);
```

AndyZe, 2023-03-17:
OK, I got it working   :)

```
      ruckig::Trajectory<ruckig::DynamicDOFs, ruckig::StandardVector> ruckig_trajectory(num_dof);
      ruckig_result = ruckig.calculate(ruckig_input, ruckig_trajectory);
      ...
      std::vector<double> new_position(num_dof);
      std::vector<double> new_velocity(num_dof);
      std::vector<double> new_acceleration(num_dof);
      ruckig_trajectory.at_time(time_from_start, new_position, new_velocity, new_acceleration);
```

closed 2023-03-17 by AndyZe

pantor, 2023-03-19:
Yes, this looks good. I've now added the sentence

> This switches the default Vector from the `std::array` to the dynamic `std::vector` type. 

to the Readme to clarify this.