pantor/ruckig #160: Random errors from time to time

hidoba, 2023-05-24:
I'm using offline trajectory computation through Python, using the version compiled from this repo from May 15, (commit eb749af10fe10554bea795182209242a9c57433a). Running on Nvidia Jetson Nano. Errors happen every once in a while.

Examples of output when errors happen:

```
ruckig.RuckigError: 
[ruckig] error in step 2 in dof: 0 for t sync: 0.000000 input: 
inp.current_position = [-0.04895883258572608, 0.1992637939208025, -0.2914660630142018]
inp.current_velocity = [1.425883388427091e-14, -9.520201039439915e-20, -9.656571194417352e-20]
inp.current_acceleration = [-2.370282711878416e-12, 0, 3.469446951953614e-18]
inp.target_position = [-0.04895883258572691, 0.1992637939208025, -0.2914660630142018]
inp.target_velocity = [0, 0, 0]
inp.target_acceleration = [0, 0, 0]
inp.max_velocity = [5, 5, 5]
inp.max_acceleration = [10, 10, 10]
inp.max_jerk = [150, 150, 150]
```

```
ruckig.RuckigError: 
[ruckig] error in step 2 in dof: 1 for t sync: 0.000000 input: 
inp.current_position = [0.03839900283757711, 0.2093483656756634, 0.03189466728432298]
inp.current_velocity = [1.466391486711824e-18, 1.924174584908152e-18, -1.542564246001646e-20]
inp.current_acceleration = [0, -3.608224830031759e-16, 0]
inp.target_position = [0.0383990028375771, 0.2093483656756634, 0.03189466728432298]
inp.target_velocity = [0, 0, 0]
inp.target_acceleration = [0, 0, 0]
inp.max_velocity = [5, 5, 5]
inp.max_acceleration = [15, 15, 15]
inp.max_jerk = [200, 200, 200]
```

vertix, 2024-08-15:
I experience the similar issue
```
ruckig.RuckigError: 
[ruckig] error in step 2 in dof: 6 for t sync: 0.172050 input: 
inp.current_position = [336.880689744143, 18.4550225058538, 175.3043233857315, 235.7433176441638, 326.5611932404152, 58.47109897613053, 110.7888198747244]
inp.current_velocity = [-6.21033908114654, 2.977862156505148, -3.980556401132896, 4.713115053485066, -6.660235541559918, 3.000972515821758, 6.660242722868195]
inp.current_acceleration = [57.3, 0, 7.971310902217803, -18.22070634308166, 77.42229644884236, 0, -77.42233818857001]
inp.target_position = [336.5054321289062, 18.72512054443359, 174.9912872314453, 236.0825881958008, 326.1792297363281, 58.74234771728516, 111.170783996582]
inp.target_velocity = [0, 0, 0, 0, 0, 0, 0]
inp.target_acceleration = [0, 0, 0, 0, 0, 0, 0]
inp.max_velocity = [50, 50, 50, 50, 50, 50, 50]
inp.max_acceleration = [57.3, 57.3, 57.3, 57.3, 570, 570, 570]
inp.max_jerk = [450, 450, 450, 450, 450, 450, 450]
inp.min_position = [-inf, -128.9, -inf, -147.8, -inf, -120.3, -inf]
inp.max_position = [inf, 128.9, inf, 147.8, inf, 120.3, inf]
```

This almost always happens closer to the end of trajectory

Neiizo, 2024-12-17:
Hi, I sometimes have the same error. Is there any update on this ?

In my current scenario, here are my variables:

``` cmd
obj.otg
-> <ruckig.Ruckig object at 0x0000013F360E7AF0>
-> special variables
-> function variables
-> degrees_of_freedom = 3
-> delta_time = 0.01
-> max_number_of_waypoints = 0
-> _pybind11_conduit_v1_ = <bound method PyCapsule._pybind11_conduit_v1_ of <ruckig.Ruckig object at 0x0000013F360E7AF0>>

obj.DoF
-> 3

obj.out

-> out.new_position = [1059.317036088455, 836.575680764959, 0]
-> out.new_velocity = [-151.9782841623022, -3733.520461186207, 0]
-> out.new_acceleration = [0, -25399.19073985847, 0]
-> out.new_jerk = [0, 400000, 0]
-> out.time = [0.01]
-> out.calculation_duration = [6.8]

obj.inp

-> inp.current_position = [1059.317036088455, 836.575680764959, 0]
-> inp.current_velocity = [-151.9782841623022, -3733.520461186207, 0]
-> inp.current_acceleration = [0, -25399.19073985847, 0]
-> inp.target_position = [1021.675160208679, 80.73646264290286, 0]
-> inp.target_velocity = [0, 0, 0]
-> inp.target_acceleration = [0, 0, 0]
-> inp.max_velocity = [2500, 5000, 5000]
-> inp.max_acceleration = [20000, 40000, 40000]
-> inp.max_jerk = [250000, 400000, 400000]
```

and my errror : 
```cmd 
File "C:\X\mastr-bot\script\scripts\compute_trajectory.py", line 57, in updateRuckig
    obj.moveStatus = obj.otg.update(obj.inp, obj.out)
  File "C:\X\mastr-bot\script\main.py", line 74, in updateSliders
    traj.updateRuckig(s)
  File "C:\X\mastr-bot\script\main.py", line 116, in update
    updateSliders(sliders, conveyors, beams)
  File "C:\X\mastr-bot\script\main.py", line 263, in <module> (Current frame)
    update()
ruckig.RuckigError: 
[ruckig] error in step 1, dof: 1 input: 
inp.current_position = [1059.317036088455, 836.575680764959, 0]
inp.current_velocity = [-151.9782841623022, -3733.520461186207, 0]
inp.current_acceleration = [0, -25399.19073985847, 0]
inp.target_position = [1021.675160208679, 80.73646264290286, 0]
inp.target_velocity = [0, 0, 0]
inp.target_acceleration = [0, 0, 0]
inp.max_velocity = [2500, 5000, 5000]
inp.max_acceleration = [20000, 40000, 40000]
inp.max_jerk = [250000, 400000, 400000]

```