pantor/ruckig #161: ./example-3_waypoints  Segmentation fault (core dumped)

dbdxnuliba, 2023-06-06:
dell@dell:~/ws_ruckig/build/ruckig$ ./example-3_waypoints 
t | p1 | p2 | p3
[ruckig] calculate trajectory via online API server.
Segmentation fault (core dumped)

pantor, 2023-06-06:
Thanks for spotting this. It seems that this segmentation fault is thrown if no internet connection is available while trying to reach our API server.

I've now added a check for this in the recent commit on the master branch. Can you try again with this one?

dbdxnuliba, 2023-06-10:
is the remote api not opensourced

pantor, 2023-06-11:
The algorithm for intermediate waypoints is not open source, but is part of the (commercial) Ruckig Pro version that also keeps this open source project going.

Just to be precise, the interface itself is open source and available [here](http://api.ruckig.com/docs).

closed 2023-06-11 by pantor

dbdxnuliba, 2023-06-13:
ok ，how much the  (commercial) Ruckig Pro version

pantor, 2023-06-14:
Thanks for your interest! Please send us a quick mail at [info@ruckig.com](mailto:info@ruckig.com), and we will get back to you there.

dbdxnuliba, 2023-06-20:
ok ，thanks