pantor/ruckig #171: Tutorial plot failure

rtkg, 2023-10-09:
The first Python example fails when [trying to plot](https://github.com/pantor/ruckig/blob/a41ce343d56f74f0e1fe41702d7206f02b21f16c/examples/01_position.py#L47) with following error:

`AttributeError: type object 'Plotter' has no attribute 'plot_trajectory'`

I'm running Ubuntu 22.04 with version 1.3.20 of the [Python Plotter library](https://pypi.org/project/plotter/).

pantor, 2023-10-09:
The plotter class refers to [this file](https://github.com/pantor/ruckig/blob/master/test/plotter.py) in the test directory, and not to a third-party package. For a quick solution, you could add this directory to your `PYTHONPATH`, set `sys.path` before `import plotter`, or just copy-paste the file into the examples directory.

I'll add a comment / fix to make this more clear and easy to use. Thanks for catching that!

rtkg, 2023-10-09:
Perfect, thanks. One small additional note: I think the `Path` class needs to be imported explicitly as well.

`from pathlib import Path`

pantor, 2023-10-09:
Should be fixed now.

closed 2023-10-09 by pantor