pantor/ruckig #177: How to integrate with an existing ros package?

programmeddeath1, 2024-01-07:
I am trying to include ruckig to my existing ros package that contains my own motion planning setup. I have tried to inclide ruckig libraries as well as globally install ruckig and add that to target_link_libraries, but during compilation it fails to be able find the function definitions, its only able to read the declarations from the header files.

Is there any document or straightforward steps that I can refer to be able to add ruckig to my ros package?

Thanks in advance!

pantor, 2024-01-07:
If you're in the ROS ecosystem, it's probably easiest just to use the [Ruckig ROS package](https://index.ros.org/p/ruckig). Otherwise you can also include Ruckig directly via catkin - more infos are [here](https://github.com/pantor/ruckig/issues/87).

pantor, 2024-01-08:
And of course there is also the CMake example [here](https://github.com/pantor/ruckig/blob/main/examples/CMakeLists-installed.txt).

programmeddeath1, 2024-01-08:
I am trying to run on ros1-noetic which doesn't have an official ruckig package in the ros-index. It is giving conflicts in terms of building and there are ros-noetic-ruckig packages installed in the desktop installed version, even though there is no documentation for it nor an official package.

I tried the catkin method as shown in that [issue](https://github.com/pantor/ruckig/issues/87)  but it is not able to reference the .cpp files in src. I did make install on ruckig for system install, but it fails to understand the .cpp header definitions 
```
In file included from /home/planners/ruckig/src/motion_planner/include/ruckig/profile.hpp:12,
                 from /home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:9,
                 from /home/planners/ruckig/src/motion_planner/include/ruckig/calculator_target.hpp:11,
                 from /home/planners/ruckig/src/motion_planner/include/ruckig/calculator.hpp:3,
                 from /home/planners/ruckig/src/motion_planner/include/ruckig/ruckig.hpp:13,
                 from /home/planners/ruckig/src/motion_planner/src/IK_FK_Tester.cpp:6:
/home/planners/ruckig/src/motion_planner/include/ruckig/roots.hpp: In function ‘double ruckig::roots::poly_eval(const std::array<double, N>&, double)’:
/home/planners/ruckig/src/motion_planner/include/ruckig/roots.hpp:295:8: warning: ‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’
  295 |     if constexpr (N == 0) {
      |        ^~~~~~~~~

=================
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:57:10: error: ‘optional’ in namespace ‘std’ does not name a template type
   57 |     std::optional<Interval> a, b;
      |          ^~~~~~~~
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:57:5: note: ‘std::optional’ is only available from C++17 onwards
   57 |     std::optional<Interval> a, b;
      |     ^~~
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp: In member function ‘void ruckig::Block::set_min_profile(const ruckig::Profile&)’:
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:49:9: error: ‘a’ was not declared in this scope
   49 |         a = std::nullopt;
      |         ^
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:49:18: error: ‘nullopt’ is not a member of ‘std’
   49 |         a = std::nullopt;
      |                  ^~~~~~~
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:50:9: error: ‘b’ was not declared in this scope
   50 |         b = std::nullopt;
      |         ^
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:50:18: error: ‘nullopt’ is not a member of ‘std’
   50 |         b = std::nullopt;
      |                  ^~~~~~~
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp: In static member function ‘static bool ruckig::Block::calculate_block(ruckig::Block&, std::array<ruckig::Profile, N>&, size_t)’:
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:76:16: warning: ‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’
   76 |             if constexpr (numerical_robust) {
      |                ^~~~~~~~~
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:81:23: error: ‘class ruckig::Block’ has no member named ‘a’
   81 |                 block.a = Interval(valid_profiles[idx_min], valid_profiles[idx_else_1]);
      |                       ^
/home/planners/ruckig/src/motion_planner/include/ruckig/block.hpp:112:19: error: ‘class ruckig::Block’ has no member named ‘a’
  112 |             
```
Im getting this and other such scope errors which i believe is the .cpp definitions are not accessible

Is there any reference for using with ros1? I apologise if the question is basic am fairly new to building separately with cmake so i'm not sure how to go about this, max that I have done is build catkin dependencies for ros.

pantor, 2024-01-09:
Ah, I see. The error you posted is not related to not finding the source files of Ruckig, instead this is because of missing support for C++17. Can you tell your compiler to enable support for C++17?

The easiest way to spot this is by this warnings: `‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’`.

Greenox-ai, 2024-01-10:
Hey I have set set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)

Greenox-ai, 2024-01-10:
I added the following lines in my CMakelists 
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
as well as built and installed ruckig with  -DCMAKE_CXX_STANDARD=17
It doesn't give the C++17 error anymore,
When calling     
Ruckig<3> otg(0.01); // control cycle
    InputParameter<3> input;
    OutputParameter<3> output;

from my file to test it , the catkin_make gives this error 
```
In file included from /usr/include/ruckig/calculator.hpp:5,
                 from /usr/include/ruckig/ruckig.hpp:13,
                 from /home/planners/ruckig/src/motion_planner/src/IK_FK_Tester.cpp:6:
/usr/include/ruckig/calculator_cloud.hpp:12:10: fatal error: json/json.hpp: No such file or directory
   12 | #include <json/json.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [motion_planner/CMakeFiles/IK_FK_Tester.dir/build.make:63: motion_planner/CMakeFiles/IK_FK_Tester.dir/src/IK_FK_Tester.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2624: motion_planner/CMakeFiles/IK_FK_Tester.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
This header seems to be imported only in the cloud_client file. Where to give reference to this? I tried removing the CLOUD_CLIENT build and compilations fro mthe ruckig CMakeLists.txt and trying to build it but, i still get the same error.

programmeddeath1, 2024-01-10:
I added the following lines in my CMakelists
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
as well as built and installed ruckig with -DCMAKE_CXX_STANDARD=17
It doesn't give the C++17 error anymore,
When calling
Ruckig<3> otg(0.01); // control cycle
InputParameter<3> input;
OutputParameter<3> output;

from my file to test it , the catkin_make gives this error
```

Explain
In file included from /usr/include/ruckig/calculator.hpp:5,
                 from /usr/include/ruckig/ruckig.hpp:13,
                 from /home/planners/ruckig/src/motion_planner/src/IK_FK_Tester.cpp:6:
/usr/include/ruckig/calculator_cloud.hpp:12:10: fatal error: json/json.hpp: No such file or directory
   12 | #include <json/json.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [motion_planner/CMakeFiles/IK_FK_Tester.dir/build.make:63: motion_planner/CMakeFiles/IK_FK_Tester.dir/src/IK_FK_Tester.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2624: motion_planner/CMakeFiles/IK_FK_Tester.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
This header seems to be imported only in the cloud_client file. Where to give reference to this? I tried removing the CLOUD_CLIENT build and compilations from the ruckig CMakeLists.txt and trying to build it but, i still get the same error.

pantor, 2024-01-10:
I'd recommend to follow this CMake example file:
```
# A CMakeLists.txt to show how to use Ruckig with CMake when installed via `make install` first.

cmake_minimum_required(VERSION 3.10)

project(ruckig_examples)

find_package(ruckig REQUIRED)

# Build the position example
add_executable(example-position examples/01_position.cpp)
target_compile_features(example-position PUBLIC cxx_std_17)

target_link_libraries(example-position PRIVATE ruckig::ruckig)
```

programmeddeath1, 2024-01-10:
Hi thank you for your quick response! 
I am following the same CMake Example please check my Cmake file below - 
```
cmake_minimum_required(VERSION 3.0.2)
project(motion_planner)
  set(CMAKE_CXX_STANDARD 17)
  set(CMAKE_CXX_STANDARD_REQUIRED ON)
  set(CMAKE_CXX_EXTENSIONS OFF)

find_package(catkin REQUIRED COMPONENTS
  rospy
  roscpp
  std_msgs
  custom_msg
  sensor_msgs
  geometry_msgs
  trajectory_msgs
  visualization_msgs
)

# Find the Ruckig library
find_package(ruckig REQUIRED)

catkin_package()

# Include directories
include_directories(
  ${catkin_INCLUDE_DIRS}
  include

)

add_executable(Motion_Planner src/Motion_Planner.cpp)
target_compile_features(Motion_Planner PUBLIC cxx_std_17)
target_link_libraries(Motion_Planner PUBLIC ${catkin_LIBRARIES} ruckig::ruckig)
add_dependencies(Motion_Planner ${catkin_EXPORTED_TARGETS})

add_executable(IK_FK_Tester src/IK_FK_Tester.cpp)
target_compile_features(IK_FK_Tester PUBLIC cxx_std_17)
target_link_libraries(IK_FK_Tester ${catkin_LIBRARIES} ruckig::ruckig )
add_dependencies(IK_FK_Tester ${catkin_EXPORTED_TARGETS})
```
It isnt able to build for the cloud.hpp and fails as below 
```
In file included from /usr/include/ruckig/calculator.hpp:5,
                 from /usr/include/ruckig/ruckig.hpp:13,
                 from /home/planners/ruckig/src/motion_planner/src/IK_FK_Tester.cpp:6:
/usr/include/ruckig/calculator_cloud.hpp:12:10: fatal error: json/json.hpp: No such file or directory
   12 | #include <json/json.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [motion_planner/CMakeFiles/IK_FK_Tester.dir/build.make:63: motion_planner/CMakeFiles/IK_FK_Tester.dir/src/IK_FK_Tester.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2624: motion_planner/CMakeFiles/IK_FK_Tester.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 91%] Linking CXX executable /home/planners/ruckig/devel/lib/motion_planner/test_client2
[ 91%] Built target test_client2
[ 92%] Linking CXX executable /home/planners/ruckig/devel/lib/motion_planner/test_publisher
[ 92%] Built target test_publisher
[ 94%] Linking CXX executable /home/planners/ruckig/devel/lib/motion_planner/Speed_Acceleration_Limit
[ 94%] Built target Speed_Acceleration_Limit
[ 95%] Linking CXX executable /home/planners/ruckig/devel/lib/motion_planner/test_client
[ 95%] Built target test_client
[ 96%] Linking CXX executable /home/planners/ruckig/devel/lib/motion_planner/RVIZ_Visualizer
[ 96%] Built target RVIZ_Visualizer
[ 97%] Linking CXX executable /home/planners/ruckig/devel/lib/motion_planner/Jogging
[ 97%] Built target Jogging
[ 98%] Linking CXX executable /home/planners/ruckig/devel/lib/motion_planner/Motion_Planner
[ 98%] Built target Motion_Planner
make: *** [Makefile:141: all] Error 2
Invoking "make -j16 -l16" failed
```

ajay01994, 2024-01-10:
Hi @pantor  any help would be appreciated. 

Thanks in advance

programmeddeath1, 2024-01-11:
I apologize it seems to be a very silly negligence from my end. I read there are no dependencies for the installation, i made the same assumption for separate package building too. 

Putting this here for any novice developers - 
I saw theres a thirdy_party library containing the required packages and moved the third_party folder from the github to the /usr/include path, now its working. 

Thank you for your help!

closed 2024-01-11 by programmeddeath1