pantor/ruckig #181: can not pip install ruckig

Blaise-wig, 2024-02-19:
I get this error when pip installing ruckig:

(DroneProject) C:\Users\Victus\anaconda3\envs\DroneProject>pip install ruckig
Collecting ruckig
  Using cached ruckig-0.9.2.tar.gz (289 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: ruckig
  Building wheel for ruckig (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\Victus\anaconda3\envs\DroneProject\python.exe' 'C:\Users\Victus\anaconda3\envs\DroneProject\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\Victus\AppData\Local\Temp\tmp6ygjzuis'
       cwd: C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490
  Complete output (95 lines):
  running bdist_wheel
  running build
  running build_ext
  -- Building for: Visual Studio 14 2015
  -- The CXX compiler identification is MSVC 19.0.24215.1
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/cl.exe - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Found Python: C:/Users/Victus/anaconda3/envs/DroneProject/python.exe (found suitable version "3.6.13", minimum required is "3.6") found components: Interpreter
  -- Performing Test HAS_MSVC_GL_LTCG
  -- Performing Test HAS_MSVC_GL_LTCG - Success
  -- Found pybind11: C:/Users/Victus/anaconda3/envs/DroneProject/Lib/site-packages/pybind11/include (found version "2.11.1")
  -- Configuring done (5.8s)
  -- Generating done (0.0s)
  CMake Warning:
    Manually-specified variables were not used by the project:

      EXAMPLE_VERSION_INFO
      PYTHON_EXECUTABLE

  -- Build files have been written to: C:/Users/Victus/AppData/Local/Temp/pip-install-mgh2_2at/ruckig_8d43405090b24a4ab7d6ca3e808e7490/build/temp.win32-3.6/Release
  Microsoft (R) Build Engine version 14.0.25420.1
  Copyright (C) Microsoft Corporation. All rights reserved.

  C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppBuild.targets(400,5): warning MSB8029: The Intermediate directory or Output directory cannot reside under the Temporary directory as it could lead to issues with incremental build. [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ZERO_CHECK.vcxproj]
    Checking Build System
  C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppBuild.targets(400,5): warning MSB8029: The Intermediate directory or Output directory cannot reside under the Temporary directory as it could lead to issues with incremental build. [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
    Building Custom Rule C:/Users/Victus/AppData/Local/Temp/pip-install-mgh2_2at/ruckig_8d43405090b24a4ab7d6ca3e808e7490/CMakeLists.txt
    brake.cpp
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(11824): warning C4346: 'std::is_same<IterImpl,nlohmann::detail::iter_impl<std::conditional<std::is_const<_Ty>::value,std::remove_const<_Ty>::type,const BasicJsonType>::type>>::value': dependent name is not a type [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
    C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(11824): note: prefix with 'typename' to indicate a type
    C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(12104): note: see reference to class template instantiation 'nlohmann::detail::iter_impl<BasicJsonType>' being compiled
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(11824): error C2061: syntax error: identifier '?$enable_if_t@$Q?AVstd:?$is_same@$RCAAB@V?$iter_impl@$RBAAB@@detail@nlohmann@@:value:((((*(std:?$is_same@$RCAAB@V?$iter_impl@$RBAAB@@detail@nlohmann@@:value:))||(*(std:?$is_same@$RCAAB@V?$iter_impl@$std:?$conditional@$Q?AVstd:?$is_const@$RBAAB@:value:(std:?$is_const@$RBAAB@:value:)$std:?$remove_const@$RBAAB@:type:$$CBRBAAB@:type:@detail@nlohmann@@:value:))))$$T' [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(11860): warning C4346: 'std::is_same<IterImpl,nlohmann::detail::iter_impl<std::conditional<std::is_const<_Ty>::value,std::remove_const<_Ty>::type,const BasicJsonType>::type>>::value': dependent name is not a type [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
    C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(11860): note: prefix with 'typename' to indicate a type
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(11860): error C2061: syntax error: identifier '?$enable_if_t@$Q?AVstd:?$is_same@$RCAAB@V?$iter_impl@$RBAAB@@detail@nlohmann@@:value:((((*(std:?$is_same@$RCAAB@V?$iter_impl@$RBAAB@@detail@nlohmann@@:value:))||(*(std:?$is_same@$RCAAB@V?$iter_impl@$std:?$conditional@$Q?AVstd:?$is_const@$RBAAB@:value:(std:?$is_const@$RBAAB@:value:)$std:?$remove_const@$RBAAB@:type:$$CBRBAAB@:type:@detail@nlohmann@@:value:))))$$T' [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(16737): warning C4346: 'std::is_same<NumberType,BasicJsonType::binary_t::value_type>::value': dependent name is not a type [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
    C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(16737): note: prefix with 'typename' to indicate a type
    C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(17009): note: see reference to class template instantiation 'nlohmann::detail::serializer<BasicJsonType>' being compiled
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(16737): error C2061: syntax error: identifier '?$enable_if_t@$Q?AVstd:?$is_integral@$RCAAB@:value:((((*(std:?$is_integral@$RCAAB@:value:))||(*(std:?$is_same@$RCAAB@$BasicJsonType@BAAB@:number_unsigned_t::value:)))||(*(std:?$is_same@$RCAAB@$BasicJsonType@BAAB@:number_integer_t::value:)))||(*(std:?$is_same@$RCAAB@$BasicJsonType@BAAB@:binary_t:value_type::value:)))H' [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\third_party\json/json.hpp(17251): fatal error C1083: Cannot open include file: 'string_view': No such file or directory [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
    position-step1.cpp
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\include\ruckig/ruckig.hpp(10): fatal error C1083: Cannot open include file: 'optional': No such file or directory [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\build\temp.win32-3.6\Release\ruckig.vcxproj]
    position-step2.cpp
  C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d43405090b24a4ab7d6ca3e808e7490\include\ruckig/block.hpp(6): fatal error C1083: Cannot open include file: 'optional': No such file or directory [C:\Users\Victus\AppData\Local\Temp\pip-install-mgh2_2at\ruckig_8d434 [... cut ...]
      subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
    File "C:\Users\Victus\anaconda3\envs\DroneProject\lib\subprocess.py", line 311, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for ruckig
Failed to build ruckig
ERROR: Could not build wheels for ruckig, which is required to install pyproject.toml-based projects

Can anyone help?

pantor, 2024-02-20:
Ruckig is using several features of C++17 (e.g. `optional` or `enable_if` that show up in your error log) as stated in the Readme [here](https://github.com/pantor/ruckig?tab=readme-ov-file#development). Please update your Visual Studio to a version that supports C++17, or [contact us](info@ruckig.com) for Ruckig versions that are compatible with C++11.

Blaise-wig, 2024-02-20:
Thank you for your response.  I am interested in the version of ruckig compatible with c++11.

Blaise-wig, 2024-02-20:
How can I get it?

pantor, 2024-02-20:
Please send us a short message at info@ruckig.com.

closed 2024-02-20 by pantor