pantor/ruckig #186: filter_intermediate_positions documentation missing/lacking

dsandber, 2024-04-08:
1) In the Ruckig documenation, where it says "Please find more information about these limitations [here]", the "here" link is broken

2) I had guessed that the parameter to filter_intermediate_positions specified how far the positions could be moved, however, while testing I saw the case where a parameter of 0.001 leads to a great position error (vs the intermediate waypoint positions) than specifying 0.01.   If this is possible, what exactly does that parameter mean?

Thanks!

-Dan

dsandber, 2024-04-08:
Dug into the source-code a bit and seems like this function doesn't move the waypoint positions, rather it eliminates ones that are under the threshold distance -- is this right?

pantor, 2024-07-25:
Sorry for the late response - yes you're assumption is correct.

I've added a short description to our docs [here](https://docs.ruckig.com/classruckig_1_1Ruckig.html#ab65d4397e0b244bbaa2cb3541e45707e).

closed 2024-07-25 by pantor