pantor/ruckig #192: Runtime Error on aarch64 System

roboticszeng, 2024-05-24:
Hello,

I'm experiencing a runtime error when attempting to execute the ruckig repository on an aarch64 system. The repository compiles successfully, but when I run the example-01 test case, I encounter the following error:

`./example-01_position: symbol lookup error: ./example-01_position: undefined symbol: _ZN6ruckig11CloudClientC1Ev`
I have not made any modifications to the code and am running the test case as is.

Here are the details of my system:

`uname -a` output:
`Linux ubuntu 5.10.120-tegra #1 SMP PREEMPT Tue Aug 1 12:32:50 PDT 2023 aarch64 aarch64 aarch64 GNU/Linux`
I would appreciate it if you could provide guidance or a fix for this issue. Thank you for your time and assistance.

Best regards!

pantor, 2024-05-27:
The compiler seems to have failed to pick up the `cloud_client.cpp` sources. Can you check that [this line](https://github.com/pantor/ruckig/blob/dae82835ec043a54ed6bc775f0776e1a7ff99124/CMakeLists.txt#L50) is executed by CMake. Do you have disabled the `BUILD_CLOUD_CLIENT ` variable?

roboticszeng, 2024-06-01:
i dont change any line of this CMake, just build follow the step in README.md

roboticszeng, 2024-06-01:
program crash in this line:
while (otg.update(input, output) == ruckig::Result::Working)

JornJorn, 2024-07-10:
@roboticszeng if you still have this issue, you could try my solution propused in #198 .
I had a similar issue, also an issue with a reference to ruckig::CloudClient.

pantor, 2024-09-06:
Sorry for the late response. I'd assume that this error comes from a mismatch of the header files and the linked library, e.g. you have installed ruckig system-wide, but now compile

Hi @roboticszeng - sorry for the late response. It seems that the linker is picking up a shared library that was compiled with the `BUILD_CLOUD_CLIENT=OFF` flag. Did you install Ruckig system-wide? 

You could...
- compile Ruckig yourself, e.g. by using CMake's `add_subdirectory` or following [this example](https://github.com/pantor/ruckig/blob/main/examples/CMakeLists-directory.txt).
- set the flag `BUILD_CLOUD_CLIENT=OFF` for your build - assuming you don't need that feature.

closed 2024-10-03 by pantor