pantor/ruckig #193: Acceleration Constrained motion vs jerk constrained motion

programmeddeath1, 2024-05-31:
I am running offline joint space planning for a delta robot using ruckig. For a pick and place path we are plotting the vertices and using ruckig to generate the motion plan for individual straight line segments.

When we are running the acceleration controlled motion (by not passing max jerk, current accel and target accel) the pick and place motion happens properly on our servo motors, but when we run the jerk controlled motion at the same max vel and max accel, almost at all values there is a jerk/jitter or there is a swing at the pick/place position indicating an overshoot or undershoot. At only a particular max jerk value after quite a lot of testing, we get a proper motion. These are only on specific max jerk values on which motion happens properly for a specific max velocity and max acceleration below which and above which we get overshoots and undershoots. And on some velocities we do not find any max jerk values for which proper smooth motion happens and only on extremely high jerks we ge proper path being followed but ofcourse extremely jerky motion.

Is this normal behavior or is there some issue in the way we have integrated ruckig into our motion planning setup?

The control cycle we are working with is 0.015 to 0.04 and we are using a hybrid servo drives running using a micro-controller setup.

is it possible to get the max_jerk value which ruckig uses by default in the acceleration control motion? In the latest release does the expose current jerk value show this jerk value?

AndyZe, 2024-08-09:
I'm late to comment but I'll just say, it makes sense that adding jerk limits would cause some overshoot and undershoot.

If you increase the jerk limit to something really large, like 10^4, does the overshoot/undershoot go away? If so, that would indicate that Ruckig is working properly.

AndyZe, 2024-08-09:
Overshooting is something that can occur to match the boundary condition at the target point, and there's nothing to prevent it. It's a side-effect of this package. Related feature request:  #154

programmeddeath1, 2024-08-19:
Yes as I mentioned at really large jerk limits the overshoot and undershoot goes away. Thank you for your clarification, it matches  with what we have observed in our motions!

If the jerk limit is set quite high, of course the motion is not as smooth, but path is perfect. We have also observed that tiny control cycles less than 0.01 are needed to get jerk controlled motion without much overshoot/undershoot at different specific jerk values.

Thank you for your response!

closed 2024-08-19 by programmeddeath1