pantor/ruckig #194: Result::ErrorExecutionTimeCalculation -by hualei

hubeihualei, 2024-06-06:
i test 4 dof example, i use the follow data， but  result=Result::ErrorExecutionTimeCalculation

input.current_position = {2.990056872367862, 2.51504516601562, 0, 0};
    input.current_velocity = {0.0, 0, 0, 0};
    input.current_acceleration = {0.0, 0, 0, 0};
input.target_position = {2.990056872367862, 2.51504516601562, 0,719};
    input.target_velocity = {0.0, 0, 0,0};
    input.target_acceleration = {0.0, 0.0, 0,0};
input.max_velocity = {225, 600, 12499.99999999999,546.454545454545};
    input.max_acceleration = {11250, 30000, 624999.9999999997,27272.72727272725};//
input.max_jerk = {562500, 1500000,31249999.99999998,1363636.363636363};

hubeihualei, 2024-06-06:
[image]

chinahuangyong, 2024-06-14:
input size is 4, but ruckig dof set 3.

pantor, 2024-07-24:
This issue is due to numerical reasons as the input exceeds the limits explained in the [Readme section](https://github.com/pantor/ruckig?tab=readme-ov-file#tests-and-numerical-stability). Using the Ruckig Pro version with a `time_scale = 100.0` works just fine. 

The `dof: 3` output from the message above is zero-indexed, so it's the last axis. Btw, you can format the code via markdown, this helps a lot for testing and debugging issues.

closed 2024-07-24 by pantor

hubeihualei, 2024-07-24:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。