pantor/ruckig #200: Without jerk limits, `validate_input()` prints nothing yet returns false. Ruckig `update()` fails.

AndyZe, 2024-08-09:
Ruckig works great for me when jerk limits are present. However I see weird behavior when no jerk limit / current acceleration / target acceleration is specified. Here's the state of the system. Seems fine as far as I can tell. All joints are within a valid range.

```
[component_container_mt-5] [ERROR] [1723213314.987161591] [moveit_servo_demo_container.moveit.core.ruckig_filter_plugin]: Ruckig jerk-limited smoothing failed with code: -100
[component_container_mt-5] [INFO] [1723213314.987237738] [moveit_servo_demo_container.moveit.core.ruckig_filter_plugin]: 0.01
[component_container_mt-5] Ruckig input:
[component_container_mt-5] 
[component_container_mt-5] inp.current_position = [0, -0.785, 0, -2.356, 0, 1.571, 0.785]
[component_container_mt-5] inp.current_velocity = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] inp.current_acceleration = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] inp.target_position = [6.299675467958987e-14, -0.7692521485658494, -1.22995790864666e-13, -2.346922384414514, 2.207760663970184e-14, 1.577670235848665, 0.7850000000000851]
[component_container_mt-5] inp.target_velocity = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] inp.target_acceleration = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] inp.max_velocity = [2.175, 2.175, 2.175, 2.175, 2.61, 2.61, 2.61]
[component_container_mt-5] inp.max_acceleration = [15, 7.5, 10, 12.5, 15, 20, 20]
[component_container_mt-5] inp.max_jerk = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] 
[component_container_mt-5] Ruckig output:
[component_container_mt-5] 
[component_container_mt-5] out.new_position = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] out.new_velocity = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] out.new_acceleration = [0, 0, 0, 0, 0, 0, 0]
[component_container_mt-5] out.time = [0]
[component_container_mt-5] out.calculation_duration = [5.384967072957742e-310]
```

When I try to run `validateInput()` the results are a little strange. Try/catch does not print anything!

```
  try
  {
    ruckig_->validate_input(*ruckig_input_, true, true);
  }
  catch (const std::runtime_error& error)
  {
    RCLCPP_ERROR_STREAM(getLogger(), "Invalid Ruckig input. " << error.what() << std::endl);
  }
```

There is a `false` boolean returned, but that doesn't give me much info to debug:

```
  if (!ruckig_->validate_input(*ruckig_input_, true, true))
  {
    std::cerr << "Invalid input!" << std::endl;
  }
```

This is for a MoveIt2 PR btw.

pantor, 2024-08-12:
Two quick responses:

1. There is a `RuckigError` exception type that you need to catch.
2. Do you mean to have all zero jerk, or do you want to have infinite jerk limits?

AndyZe, 2024-08-12:
>     2. Do you mean to have all zero jerk, or do you want to have infinite jerk limits?

I'm trying to do what's mentioned in the README:

> If you only want to have a acceleration-constrained trajectory, you can also omit the max_jerk as well as the current and target_acceleration value.

Not clear if that means I need to clear the vectors (current accel, target accel, max jerk) or just leave them as they were initialized. Doesn't seem to work either way.

Good point though, I could fill `max_jerk` with DBL_MAX and effectively get the same thing.

Aadi0902, 2024-11-13:
@pantor I tested the release V0.14.0 (installed using `apt-get install ros-humble-ruckig`). If we ignore initilaizing current acceleration, target acceleration, and max jerk, they were all initialized to 0s and thus when running `otg.calculate()` for offline trajectory generation, it gives out `ErrorInvalidInput` error tag.

pantor, 2024-11-14:
@Aadi0902 I'm not sure if I get that - please provide a minimum example to reproduce the issue.
- You can turn on `Ruckig`'s `throw_error` template parameter to get details about the invalid input error.  
- As stated in the readme, in general Ruckig expects the complete current state, target state and limits. Some of the InputParameters have default values: 0s for `current_velocity`, `current_acceleration`, `target_velocity`, `target_acceleration` and infs for `max_acceleration`, and `max_jerk`. You need to provide at least: `current_position`, `target_position`, and `max_velocity`.

pantor, 2024-12-08:
Feel free to reopen if you have further questions.

closed 2024-12-08 by pantor