pantor/ruckig #202: default repository checkout in install workspace fails to include nlohmann dependency

v4hn, 2024-09-04:
Hi @pantor,

thank you for addressing issues related to this before, but I'm afaid the full issue is not resolved yet.
Concretely, trying to set up another [extensive ROS-O builder](https://v4hn.github.io/ros-o-builder/), I currently experience the problem that default `ruckig` sets `BUILD_CLOUD_CLIENT=ON`, but does not ensure that [`nlohmann/json.hpp`](https://github.com/pantor/ruckig/blob/801c5f6bab618388f4383d4da3be49022bcba692/include/ruckig/calculator_cloud.hpp#L12) is available after installation.
The easiest way to fix this would be to add a `<build_export_depend>nlohmann-json-dev</build_export_depend>` in general and not use it when the cloud client is disabled, or provide a separate cmake target that sets the compile flag.

As we explicitly not use the webclient in MoveIt I just proposed to manually drop the compile flag from the exposed target to avoid the unnecessary (/undeclared) dependency. This essentially mirrors a second cmake target on the user side.

Cross-reference https://github.com/moveit/moveit/pull/3636

pantor, 2024-09-05:
PR #203 installs the missing headers into `include/ruckig/third_party`.  I'm not a big fan of bundling the dependencies into the package, but (1) I also want to support older platforms and (2) it's only two header files. WDYT?

pantor, 2024-12-08:
Feel free to reopen if you still see any problems with the current solution.

closed 2024-12-08 by pantor