pantor/ruckig #216: Array index out of bound

mskripnik, 2024-12-19:
I get sometimes "array index out of bound" exception for very short trajectories. Here's what I've found in the debugger:
[image]
Might this be a numerical issue not properly handled in Ruckig?

mskripnik, 2024-12-19:
Here is the code to reproduce:
```
int main()
{
    ruckig::Ruckig<2> ruckig;
    ruckig::InputParameter<2> ruckig_input;
    ruckig::Trajectory<2> trajectory;

    double current_px = 446.62116880136932;
    double current_py = 42.735729882548974;
    double current_vx = 4.4408920985006262e-16;
    double current_vy = -1.6205012520938689e-18;
    double current_ax = 0;
    double current_ay = -4.0512531302346743e-17;

    double target_px = 446.62116880136932;
    double target_py = 42.735729882548974;
    double target_vx = 0;
    double target_vy = 0;
    double target_ax = 0;
    double target_ay = 0;

    double v_max = 500;
    double a_max = 1000 * 0.5;
    double j_max = 10000 * 0.5;

    ruckig_input.current_position     = {current_px, current_py};
    ruckig_input.current_velocity     = {current_vx, current_vy};
    ruckig_input.current_acceleration = {current_ax, current_ay};

    ruckig_input.target_position     = {target_px, target_py};
    ruckig_input.target_velocity     = {target_vx, target_vy};
    ruckig_input.target_acceleration = {target_ax, target_ay};

    ruckig_input.max_velocity     = {v_max, v_max};
    ruckig_input.max_acceleration = {a_max, a_max};
    ruckig_input.max_jerk         = {j_max, j_max};

    ruckig_input.control_interface = ruckig::ControlInterface::Position;
    ruckig_input.synchronization   = ruckig::Synchronization::Phase;

    ruckig.calculate(ruckig_input, trajectory);

    array<double, 2> p, v, a;
    trajectory.at_time(0, p, v, a);

    return 0;
}
```

pantor, 2024-12-19:
Thanks for reporting! 

This (and also other issues like #215) are of course fixed in the Pro version. I'll still keep this open for reference.