pantor/ruckig #229: Python binding of `Trajectory.get_first_time_at_position` always returns `0.0` or `None`

j-m-jiang, 2025-05-08:
After upgrading the Python `ruckig` package to version `0.12.2` or `0.14.0`, calling `Trajectory.get_first_time_at_position` will always result in `0.0` if the queried position is inside the extrema. This seems to be caused by L80-86 of `python.cpp` not updated for the new signature of `Trajectory.get_first_time_at_position` in the C++ end.

pantor, 2025-05-09:
Do you have an example to reproduce this issue?

j-m-jiang, 2025-05-09:
Minimum example:
```
## ruckig_example.py
import ruckig
from ruckig import InputParameter, Ruckig, Trajectory

print(f'{ruckig.__version__=}\n')

ruckig_1 = Ruckig(1)
traj = Trajectory(1)

input_params = InputParameter(1)

input_params.current_position = [0.0]
input_params.current_velocity = [0.0]
input_params.current_acceleration = [0.0]
input_params.target_position = [1.0]
input_params.target_velocity = [0.0]
input_params.target_acceleration = [0.0]
input_params.max_velocity = [1.0]
input_params.max_acceleration = [2.0]
input_params.max_jerk = [3.0]

assert ruckig_1.validate_input(
    input_params,
    check_current_state_within_limits=True,
    check_target_state_within_limits=True,
)

ruckig_1.calculate(input_params, traj)

print(f'{traj.duration=}\n')

print(f'{traj.get_first_time_at_position(0, -0.1)=}')
print(f'{traj.get_first_time_at_position(0, 0.0)=}')
print(f'{traj.get_first_time_at_position(0, 0.5)=}')
print(f'{traj.get_first_time_at_position(0, 1.0)=}')
print(f'{traj.get_first_time_at_position(0, 1.1)=}\n')

print(f'{traj.at_time(0.0)=}')
print(f'{traj.at_time(traj.duration/2.0)=}')
print(f'{traj.at_time(traj.duration)=}')
```

Output:
```
> python ruckig_example.py 
ruckig.__version__='0.14.0'

traj.duration=2.201284832596418

traj.get_first_time_at_position(0, -0.1)=None
traj.get_first_time_at_position(0, 0.0)=0.0
traj.get_first_time_at_position(0, 0.5)=0.0
traj.get_first_time_at_position(0, 1.0)=0.0
traj.get_first_time_at_position(0, 1.1)=None

traj.at_time(0.0)=([0.0], [0.0], [0.0])
traj.at_time(traj.duration/2.0)=([0.5], [0.9085602964160698], [0.0])
traj.at_time(traj.duration)=([1.0], [0.0], [0.0])
```

pantor, 2025-05-10:
Does this still happen when you install the main branch of ruckig via `pip install git+https://github.com/pantor/ruckig.git`?

j-m-jiang, 2025-05-10:
Looks like it is indeed fixed on `main`!

I can pin my requirements to this specific commit for now, but would it be possible to get this version as a bug fix release? Thank you very much!

pantor, 2025-05-16:
v0.15.3 is now on pypi.